magicbug/Cloudlog #651: Suggestion - missing country files cause SQL errors

philpem, 2020-10-09:
**Is your feature request related to a problem? Please describe.**

When creating the first station profile in a new installation, I got an SQL error. This was due to missing the "Update Country Files" step in the installation guide. (I ran through the guide a second time and spotted it)

**Describe the solution you'd like**

Instead of a confusing SQL error, I feel it would be more helpful if an empty Country File database resulted in an error which clearly spelled out the problem and solution, e.g. "click Update Country Files".

**Describe alternatives you've considered**

- Make DXCC data download part of the installation process (an extra step before "the installation is complete, delete the 'install' directory")
- A warning or hint on the front page of a new installation would be helpful -- indicating that the DXCC country file data isn't present and needs to be installed.

**Additional context**

Steps to reproduce the behaviour:
1. Install Cloudlog per https://github.com/magicbug/Cloudlog/wiki/Installation - but skip downloading DXCC data.
2. Admin -> Station Profiles - create station profile
3. Fill in the form and submit it (note that the only hint that country file data is missing is that the DXCC form field isn't present)
4. See error:

```
A Database Error Occurred

Error Number: 1366

Incorrect integer value: '' for column `cloudlog`.`station_profile`.`station_dxcc` at row 1

INSERT INTO `station_profile` (`station_profile_name`, `station_gridsquare`, `station_city`, `station_iota`, `station_sota`, `station_callsign`, `station_dxcc`, `station_country`, `station_cnty`, `station_cq`, `station_itu`, `state`, `eqslqthnickname`, `qrzapikey`) VALUES ('Home (M0)', 'IO93GT', 'Leeds', '', '', 'M0OFX', '', '', '', '1', '1', '', '', '')

Filename: models/Stations.php

Line Number: 57
```

dg9vh, 2020-10-09:
Another idea for solving the problem would be to fill the table initially with data at installation-stage... I mean when creating the table also fill it initially with data with the installation-scripts...

magicbug, 2020-10-09:
I actually did prefill and it caused time outs too, hence the current solution.

magicbug, 2020-10-09:
The solution is to improve the install system so that you get stuck on a check list on a fresh install, so that certain parts aren't available

closed 2020-10-12 by magicbug