magicbug/Cloudlog #652: Suggestion - Normalise heading styles

poll-busily, 2020-10-10:
**Is your feature request related to a problem? Please describe.**
There are three or four styles used on the admin pages when it comes to headings and sub-text. These are the main ones:

1. Small header text, small description text. E.g Radio Interfaces page
2. As above but the first line of the description text is a large font. E.g. API page
3. Big header text, even bigger title above that, small description text. E.g ADIF page

The last style uses the big title text when the page in question has multiple parts.

**Describe the solution you'd like**
I'd like to clean these up a bit but wanted to check the preferred style before starting any work. We could pick one of the three main styles already in use or define something new.

magicbug, 2020-10-12:
@poll-busily Hi Paul sorry for taking a while to get back to you, Yes there needs to be some unification of the page titles, ideally coming off $page_title so page titles and <title> tags match.

I'd suggest we match everything to /index.php/lotw then any titles within the cards are relevant to the data in the box if that makes sense.

Oh and thanks for correcting my spelling it is appreciated and cleans up the interface

poll-busily, 2020-10-14:
That makes sense to me, yeah. I'll make a first pass and see how it looks.

PR #670 (2020-10-22): Normalise styles of admin pages
Resolves #652 , at least for pages under the Admin menu drop-down. This PR uses the LotW page style (H2 headings, title taken from $data['page_title']) with non-bold card headings and plain card description text. I tweaked the card descriptions for certain pages. For the Updates page I added a card to wrap the content as there wasn't one already.
closed 2020-10-22 by magicbug