magicbug/Cloudlog #664: Frequency Fields not Shown

myzinsky, 2020-10-18:
**Describe the bug**
When uploading via the API sometimes the frequency is not shown and the band is not correct:
[image]

magicbug, 2020-10-18:
Not enough information to debug, if the data is coming via the API, what data was sent?

magicbug, 2020-10-18:
Also useful to see an SQL export of the QSO

myzinsky, 2020-10-18:
okay the bug is twofold. As mentioned above it would be nice to show the frequency field also when it is empty. I can create a PR for this. For the Synchronisation Problem I assume its a problem of my app.

magicbug, 2020-10-18:
Happy for frequency fields to be shown on edit think it would be a useful addition

closed 2020-10-18 by magicbug