magicbug/Cloudlog #669: 'Accumulated statistics' is running very long

dg9vh, 2020-10-20:
**Describe the bug**
Running the Accumulated statistics will take very long on instances with >10000 log-entries.

**To Reproduce**
Steps to reproduce the behaviour:
1. Create log with many entries
2. run Accumulated statistics for all bands based on years
3. go shopping...

**Expected behaviour**
Calculation should run faster

**Additional context**
Maybe setting additional indices on some tables (especially log-table) could improve performance?

AndreasK79, 2020-10-20:
Hi Kim.

I know the monthly filter can be very slow, but yearly shouldn't take too long. That is why I put a loading indicator on the button. I do not know if it's possible to speed up the query itself, so perhaps some indexes might help. Any suggestions are welcome, as I'm not sure what would help here.

When testing locally on DXCC:
Yearly: 4.76s
Monthly: 46s

dg9vh, 2020-10-20:
I will take a look at the SQL you use this evening... 73 de Kim

magicbug, 2020-10-22:
Takes a while but not too terrible I think most fields are indexed that it uses

AndreasK79, 2020-12-25:
As we are stuck with and old version of MySQL, I'm not sure there is much we can do.

This query runs very fast, but needs version 8 or newer:

> SELECT distinct COUNT(col_dxcc) over (order by month), month
FROM (
  SELECT col_dxcc, MIN(date_format(col_time_on, '%Y%m')) month
  FROM table_hrd_contacts_v01 thcv
  where station_id  = 1
  group by 1
  ) as x
order by 1

magicbug, 2020-12-26:
Sadly till V5.7 becomes EOL on stable platforms it will be min for Cloudlog so I agree, very little we can do its an expensive SQL query but no real solution.

closed 2020-12-26 by magicbug

AndreasK79, 2020-12-26:
Fetching the data and doing it in code, is a possibility, but I'm not sure it's worth the time and effort. I've searched a lot on this subject, and the queries I have found, is usually using windowing functions (which came in MySQL v8). I have not found another one without, that is faster.