magicbug/Cloudlog #675: eQSL export problem on satellite QSOs

rfamp, 2020-10-27:
Hi, thank you for creating this wonderful piece of software but I have a problem logging satellite QSOs on eQSL.cc

When uploading new created satellite QSOs on eQSL, the satellite and mode is not set correctly. The satellite seems to be the eQSL.cc default setting (AO-27) and the mode is not selected. So after exporting the QSOs I have to correct them on eQSL.cc to be the right mode

Steps to reproduce the behaviour:
Log a Satellite QSO and export it to eQSL.cc

I looked inside the eqsl.php and compared it to the eqsl adif content specs and I think the main problem is, that cloudlog automatically exports  BAND and SAT_BAND which maybe confusing for eqsl but I'm not sure
There seems to be someone with the same problem on the forum (including screenshots):
https://forum.cloudlog.co.uk/d/71-upload-sat-qso-to-eqsl

Best Regards

magicbug, 2020-10-27:
I need to check into this, we send SAT_NAME and SAT_Mode these are standard ADIF fields same as we send to LoTW, but not aware of anything in the code that might cause this, will be tomorrow but its top of the list.

magicbug, 2020-10-27:
Note for future self or someone else looking before I do, in the code for the ADIF field name we have SAT_NAME for example but other fields further up use %5F instead of _ I suspect this might actually be the problem

magicbug, 2020-10-27:
I've just actually quickly changed things to reflect the above comment in the code see https://github.com/magicbug/Cloudlog/blob/master/application/controllers/Eqsl.php

I haven't tested but based on the fact its used in the code above I doubt it is breaking change, might be worth a shot with the latest code base

rfamp, 2020-10-27:
Hi, thank you for the (really) quick answer, but sadly, this change doesn't change the issue. I tried to log a QSO via SO-50, AO-91 or PO-101 and multiple modes but it keeps the same on eqsl.cc, the satellite is always AO-27 and no mode is choosen.

magicbug, 2020-10-27:
Back to the drawing board tomorrow then :)

rfamp, 2020-10-27:
I also noted that the error is already in the "eQSL QSO Upload" Dialog. The mode is FM but the 
Band is not the satellite as shown on the logbook page, it is the Band if the qso would be a "normal" QSO. 
To reproduce this, just start a log, choose. e.g 160m as a Band, enter the Date, Call, etc., switch to the Satellite Tab, enter the Satellite, the Mode and hit save.
The QSO is now listed correctly with the sat name as Band in the logbook list but when looking at the eqsl upload, the Band is 160m again. Maybe some confusion between SAT_MODE and MODE in the eqsl adif requirements?

magicbug, 2020-10-27:
Confused to what your referring to, on Cloudlog frontend I swap band for
sat name as it's more useful, however bands still set properly and stored
within the database.

rfamp, 2020-10-28:
Ok, my bad, now I mixed something up, sorry. I thought that was the root of the problem, but I was wrong. 

> Confused to what your referring to, on Cloudlog frontend I swap band for sat name as it's more useful, however bands still set properly and stored within the database.

rfamp, 2020-10-29:
I did some further investigation and compared the adif lines of two QSOs
The first one which I entered on eqsl.cc by hand:
`<CALL:5>CALLSIGN<QSO_DATE:8:D>20200101<TIME_ON:4>2145<BAND:2>2m<MODE:2>FM<RST_SENT:2>59<SAT_NAME:5>SO-50<SAT_MODE:2>VU<PROP_MODE:3>SAT<QSL_SENT:1>Y<QSL_SENT_VIA:1>E<QSLMSG:24>TNX for the qso via SO50<EOR>
`

And the second one which has been uploaded by cloudlog:
`<CALL:5>CALLSIGN<QSO_DATE:8:D>20202101<TIME_ON:4>1635<BAND:2>2M<MODE:2>FM<RST_SENT:2>59<PROP_MODE:3>SAT<QSL_SENT:1>Y<QSL_SENT_VIA:1>E<EOR>
`
So:
SAT_NAME,SAT_MODE and QSLMSG are missing

Comparing it to the eqsl.php Iine 412, I think I spotted the error:
`if ($qsl['COL_SAT_NAME'] = '')`
So the name is just included if it's empty? PROP_MODE works fine and it's  if statement is:
`if ($qsl['COL_PROP_MODE'])`

So removing the comparison in the statement and adding QSLMSG should fix it or am I wrong?

magicbug, 2020-10-29:
All fixed

closed 2020-10-29 by magicbug

rfamp, 2020-10-29:
Works like a charm, thank you!

commit ecaba076 (2020-10-29): [eQSL] Fixes issue of Sat name/mode not being sent.

Fixes #675