magicbug/Cloudlog #680: Uninitialized string offset error in Adif_parser.php on esql import

M0LTE, 2020-11-01:
**Describe the bug**
When I run eqsl import, I get the following errors:

A PHP Error was encountered
Severity: Notice
Message: Uninitialized string offset: 158
Filename: libraries/Adif_parser.php
Line Number: 159

A PHP Error was encountered
Severity: Notice
Message: Uninitialized string offset: 158
Filename: libraries/Adif_parser.php
Line Number: 165

Each one appears twice.

The rest of the page renders fine below the errors.

**To Reproduce**
Steps to reproduce the behaviour:
1. Admin -> eQSL import
2. Pull eQSL data for me
3. Import eQSL QSO matches
4. See error

I pulled from master the other day so I'm pretty up to date.

**Expected behaviour**
Error to not occur

magicbug, 2020-11-01:
I'm actually not sure what caused it, I'd suggest grabbing the ADIF file from eQSL so we can look at it, but without that cant do too much about it.

lu1aax, 2020-11-01:
Very same issue here...

magicbug, 2020-11-01:
Stupid question, but does sending to eQSL work?

M0LTE, 2020-11-01:
> Stupid question, but does sending to eQSL work?

yes, no errors

M0LTE, 2020-11-01:
Sent you a (twitter!) DM with a link to my eQSL ADIF export @magicbug

M0LTE, 2020-11-13:
Still persisting, so I guess not a transient eQSL issue

AndreasK79, 2020-11-21:
@M0LTE did Peter find anything wrong in the ADIF export from eQSL?

M0LTE, 2020-11-21:
Not heard back.

AndreasK79, 2020-11-21:
Ok. I could take a look if you send me a link to your export.

closed 2020-12-27 by magicbug