magicbug/Cloudlog #684: Wrong frequency for QO-100 QSOs

DG7YEO, 2020-11-04:
There was already a report on this together with SatPC-32 Interface which is closed but still can't get this fixed in my setup. Even with the latest update I did today still get negative value in the Frequency field.

If I edit a QSO I did via QO-100 I see a negative value in the Frequency.
When I added the Frequency in live QSO I add ie 2400000000 when added all other and save the QSO and later would like to edit the QSO Data I get a negative value in the Frequency, see picture below.

[image]

As in the other thread shown it seems to be an issue with 32 vs 64 bit, but my systems seems to be 64bit, as far as I can check. My cloudlog setup runs on my virtual webservice at my hoster domain factory.
I checked the system the  system the webserver is runnning on and it is a 64bit system, I can check the DB Server as I don't have access to it. 

I tried to change the frequency values in the database for one QSO using phpmyadmin webgui and when I view this QSO in cloudlog afterwards I shows correct values but as soon as I save the QSO again it change back to the wrong negative value.

[image]
Above you see correct values after I changed them directly in the DB, now select the RX band and click on save canges in cloudlog and open the QSO again and Frequency is wrong again, see below:
[image]

As I can set the correct Frequency directly in the DB I expect it is on a 64bit system and the uname -a on the Webserver sa it is a x86_64 GNU/Linux system this is as well a 64bit. 

Any idea what else I could check? 
73
ELmar
DG7YEO

DG7YEO, 2020-11-17:
I had a ticket with my webhoster, seems the old webhosting product is not 100% 64bit capable, I now requested an update to a new product which should be 100% 64Bit capable. When this is done I hope that issue is fixed.

magicbug, 2020-11-21:
I'm closing this issue as its not a Cloudlog issue but rather a user environmental one due to 32bit OS limitations

closed 2020-11-21 by magicbug