magicbug/Cloudlog #685: Analytics > Distances Worked not following user selected distance measurement

magicbug, 2020-11-06:
The Analytics > Distances Worked area doesn't follow the user's measurement choice should be an easy fix

daccle, 2020-11-09:
The calculation is done in `models/distance_model.php:35`:

```php
    // This functions takes query result from the database and extracts grids from the qso,
    // then calculates distance between homelocator and locator given in qso.
    // It builds an array, which has 50km intervals, then inputs each length into the correct spot
    // The function returns a json-encoded array.
    function plot($qsoArray, $gridsquare) {
        $stationgrid = strtoupper($gridsquare[0]);              // We use only the first entered gridsquare from the active profile
        if (strlen($stationgrid) == 4) $stationgrid .= 'MM';    // adding center of grid if only 4 digits are specified

        if (!$this->valid_locator($stationgrid)) {
            header('Content-Type: application/json');
            echo json_encode(array('Error' => 'Error. There is a problem with the gridsquare set in your profile!'));
        }
        else {
            // Making the array we will use for plotting, we save occurrences of the length of each qso in the array
            $j = 0;
            for ($i = 0; $j < 20000; $i++) {
                $dataarray[$i]['dist'] =  $j . 'km - ' . ($j + 50) . 'km';
                $dataarray[$i]['count'] = 0;
                $dataarray[$i]['calls'] = '';
                $dataarray[$i]['callcount'] = 0;
                $j += 50;
            }
    
            $qrb = array (					                                            // Used for storing the QSO with the longest QRB
                'Callsign' => '',
                'Grid' => '',
                'Distance' => '',
                'Qsoes' => '',
                'Grids' => ''
            );
    
            foreach ($qsoArray as $qso) {
                $qrb['Qsoes']++;                                                        // Counts up number of qsoes
                $bearingdistance = $this->bearing_dist($stationgrid, $qso['grid']);     // Calculates distance based on grids
                $arrayplacement = $bearingdistance / 50;                                // Resolution is 50, calculates where to put result in array
                if ($bearingdistance > $qrb['Distance']) {                              // Saves the longest QSO
                    $qrb['Distance'] = $bearingdistance;
                    $qrb['Callsign'] = $qso['callsign'];
                    $qrb['Grid'] = $qso['grid'];
                }
                $dataarray[$arrayplacement]['count']++;                                               // Used for counting total qsoes plotted
                if ($dataarray[$arrayplacement]['callcount'] < 5) {                     // Used for tooltip in graph, set limit to 5 calls shown
                    if ($dataarray[$arrayplacement]['callcount'] > 0) {
                        $dataarray[$arrayplacement]['calls'] .= ', ';
                    }
                    $dataarray[$arrayplacement]['calls'] .= $qso['callsign'];
                    $dataarray[$arrayplacement]['callcount']++;
                }
            }
    
            if (!$qrb['Qsoes'] == 0) {  // We have a result :)
                header('Content-Type: application/json');
                $data['ok'] = 'OK';
                $data['qrb'] = $qrb;
                $data['qsodata'] = $dataarray;
                echo json_encode($data);
            }
            else {
                header('Content-Type: application/json');
                echo json_encode(array('Error' => 'No QSOs found to plot.'));
            }
        }
    }
```

The data is stored in kilometers, so I would suggest to have a helper function which take any distance in km returns a distance string in the user choosen distance setting. I am not sure on how to do this plot. Stick with the 50km intervals or have something adapted to nautical or imperial distances?

I would try to implement it, but I need some ideas on the plot intervals.

AndreasK79, 2020-11-09:
I'm the author of this code, and the one who implemented it into Cloudlog.

I think the easiest solution would be to leave the interval as is, and change the text. Then the factor at line 157 in distance_model must be changed according to the unit used. A few modifications to the javascript must be done as well. If I remember correctly, there are a few references to km there as well. It is located in views\interface_assets\header.php.

magicbug, 2020-11-09:
Agreed this should just be a frontend change, but creating a helper that can be used across the site is a decent idea as it will be needed all over the place at times.

daccle, 2020-11-09:
Is there already a place for such global helper functions? I will need a bit of uninterupted time to figure this out, but I willing to write such a helper later this week.

daccle, 2020-11-16:
Well I tried to solve this, but currently I am unable to do this. Maybe I am missing some concepts of Cloudlog, or my days as php web dev are too long gone.
Maybe @AndreasK79 can help?

AndreasK79, 2020-11-16:
@daccle I'll see if I can figure something out here.

AndreasK79, 2020-11-20:
I'm halfway there. It's fixed in the backend. Now I need to fix it in the javascript. I also noticed that highstock is broken right now, because the map-file is not included. I will include this in the PR when it's done.

PR #715 (2020-11-21): [Distance plotting] This now follows chosen distance measurement. MAP…
… file added for broken highstock.js. Fixed #685.
closed 2020-11-21 by magicbug