magicbug/Cloudlog #688: Add My Sig and My Sig Info to station profile in order to support WWFF or similar

daccle, 2020-11-08:
I operate regulary portable in the WWFF (https://wwff.co/) contest/award, so I would like to add a WWFF reference (e.g. DLFF-0029) to my station profile. In the ADIF Export, this should be something like:

`<MY_SIG:4>WWFF <MY_SIG_INFO:8>DLFF-0029`

I am willing to implement this for myself, but I would love to have some help in planning the needed steps and I need somebody to review my code.

daccle, 2020-11-08:
I've started my work here: https://github.com/daccle/Cloudlog/tree/add_sig_to_station

So far I've edited the model and the view. W

magicbug, 2020-11-08:
You'd need to create a migration file to add the fields to station profile table then adjust the station profile model and frontend to add these with validation for whatever is expected.

I'll admit wasn't on my roadmap to add but no objection to it.

daccle, 2020-11-08:
Hi Peter!
I've added the changes to the model, the edit/create view. I've made a migration to the respektive fields into the station_profile table. Now I am struggeling to make sure, that 'sig' and 'sig_info' are added to the QSO.

The TABLE_HRD_CONTACTS_V01 already has the fields COL_MY_SIG and COL_MY_SIG_INFO but I don't know how to populate them. Any advise is kindly appreciated.

PR #690 (2020-11-16): Add sig and sig_info to station #688
This adds sig and sig_info fields to the Stations model. The given information will be part of the log in the columns COL_MY_SIG and COL_MY_SIG_INFO. This information is useful for things like WWFF.

This is my first pull request, please review in depth and please give me some feedback.

Should fix #688
closed 2020-11-16 by magicbug