magicbug/Cloudlog #697: Date preference not reflected on Live QSO or Post QSO pages

ofadam, 2020-11-13:
The date preference specified in the user profile isn't reflected correctly on the Live QSO or Post QSO pages.

magicbug, 2020-11-13:
Correct aware it's been on my to-do list but keeps nudging down the list.

dgdavid, 2021-01-11:
> The date preference specified in the user profile isn't reflected correctly on the Live QSO or Post QSO pages.

Neither in the [QSO widget](https://github.com/magicbug/Cloudlog/blob/fba3a16ecc2796e3fbfadea663381cafcdb2dd06/application/views/widgets/qsos.php#L29), as said in  https://github.com/magicbug/Cloudlog/discussions/767#discussioncomment-269505

magicbug, 2021-07-24:
> > The date preference specified in the user profile isn't reflected correctly on the Live QSO or Post QSO pages.
> 
> Neither in the [QSO widget](https://github.com/magicbug/Cloudlog/blob/fba3a16ecc2796e3fbfadea663381cafcdb2dd06/application/views/widgets/qsos.php#L29), as said in [#767 (comment)](https://github.com/magicbug/Cloudlog/discussions/767#discussioncomment-269505)

QSO Widget should be addressed

T012m3n7oR, 2021-09-22:
Live QSO - I believe this is in footer.php around [line 506](https://github.com/magicbug/Cloudlog/blob/50cd201573743cbb075c559eda51a7696175e678/application/views/interface_assets/footer.php#L506).  With this being the js, I'm not entirely sure how to convert the date format that is stored on the user profile to the js format string without writing a conversion function, or some kind of dictionary/array lookup.

Post QSO - Appears to be in index.php around [line 45](https://github.com/magicbug/Cloudlog/blob/4cfc4451e6c2d019eefabe07a0abb700b84d6ade/application/views/qso/index.php#L45).  The math that it is completing on 45/47 I don't completely understand.  When will server time ever be 24+ hours (86,400 seconds) off of UTC time?  I don't understand the use of the hidden fields between lines 53-57 either, but they appear to only be used on Live QSO and not related to this issue directly as far as I can tell.

AndreasK79, 2021-09-22:
@T012m3n7oR Yeah, that is correct. I don't know how to do this in JS, that's why I haven't fixed it. I don't know if @magicbug knows either.

If I remember correctly, the calculation that is done, is to check if the stored date is less than 24 hours old. if it is, use it, if not use current.

T012m3n7oR, 2021-09-22:
@AndreasK79 I actually just fixed this locally.  You can call PHP in the middle of the JS... so it's definitely dirty, but you can use that to look up the user's setting, and then I'm doing str_replace to replace the values to the jclock equivalents.

I've since tested a few different time formats with the saving QSO post and Live pages and they still seem to work without issue.

I'll PR probably within the next hour or two.

AndreasK79, 2021-09-22:
@T012m3n7oR guess we solved that PR problem. Great that you found a solution, even though it's a dirty fix.

closed 2022-12-01 by magicbug

T012m3n7oR, 2022-12-01:
@magicbug wasnt this reverted?