magicbug/Cloudlog #700: Feature request: support for Parks on the Air

ryandzink, 2020-11-16:
**Describe the solution you'd like**
Would it be possible to add Parks on the Air/POTA to as an awards category/field, similar to SOTA references? I think something like "POTA Number" as an input field on the QSO entry field similar to the SOTA reference would work for this.

**Describe alternatives you've considered**
Currently I am just adding POTA details to the notes in the QSO but it would be nice to have some similar Awards features like SOTA/IOTA etc.

ryandzink, 2020-11-16:
I'm going to close this as it seems to be a duplicate of or very similar to #688 . POTA uses the same MY_SIG_INFO field, with the addition of a SIG_INFO field for Park-to-Park QSOs.

closed 2020-11-16 by ryandzink

daccle, 2020-11-16:
Well, only parts of the issue are solved in #688 The enable Park2Park contacts, we would to have complementary to MY_SIG and MY_SIG_INFO, fields for SIG and SIG_INFO as well. They would need to added to the QSO dialog as well. I am willing to add this, but before I fiddle with the QSO dialog I would to have the consent of @magicbug himself.