magicbug/Cloudlog #702: Ability to update existing QSO with third-party data

ofadam, 2020-11-16:
When importing ADIF files from WSJT-X, there is no name or other information stored. Could there be a way to open up the QSO once in Cloudlog and update with data from QRZ.com?

dg9vh, 2020-11-16:
Idea could be a 'fetch qrz.com data' button filling up empty fields and fields with data could be marked with some background color and as tool tip could be shown the fetched value so the user could modify such values manually if wanted.

magicbug, 2020-11-17:
Yes agree would be a nice feature, although this would need to be presented to a user so they can decide whether they want to add the information to the QSO

jsb2092, 2020-11-18:
Is this not already part of the lookup on qrz or hamqth import feature (though it has to be setup in the config.ini).  In the feature I was adding, it does update the name on import, but only if it's not already set, same with grid square.  Having a button could be good too for those that were missed, or if the feature isn't turned on.

closed 2020-12-11 by magicbug