magicbug/Cloudlog #704: Add POTA (Parks on the Air) to the awards section

lasantiagoa, 2020-11-16:
Incidentally, when SIG_INFO and MY_SIG_INFO would need to be used when exporting those log entries.
SIG_INFO (park where the other station was operating)
MY_SIG_INFO (the park reference from where I am operating)

That way the log can be exported and sent to POTA for processing.

daccle, 2020-11-16:
Don't forget about MY_SIG and SIG. MY_SIG and MY_SIG_INFO fields are available since #688 is merged. See my comment in #700 to enable Park2Park.

daccle, 2020-11-16:
I would suggest to close this issue and start a pull request to add SIG and SIG_INFO.

daccle, 2020-11-16:
This still doesn't offer POTA in the AWARDS section, but since there are a couple of funny things based on SIG/SIG_INFO we should aim for a more general solution, since adding POTA, COTA and like WWFF on their own would crowd IMHO the AWARDS section

magicbug, 2020-11-17:
Biggest issue is theres so many awards now that it could end up with cloudlog having 20 fields just for different award types which gets a bit much, issue is I don't really know the best way to handle it.

daccle, 2020-11-17:
I am looking into the awards which are based on SIG/SIG_INFO. I have no solution so far, but I don't want to clutter CloudLog. So either there is a solution based on a config setting or we parse the content of SIG to generate a list of possible awards.

For the moment I would to see this issue here to be closed. I will come up with a proposal for a more general solution.

dg9vh, 2020-11-17:
How about having just a few fields (I think 3 should be ok) where we have a listbox for label (to choose the award programme) and a textfield to enter the value for the number? Or to make it flexible: have an award table where we can enter n awards for each qso and display them into a table in the QSO-edit-dialog? so we can have 1 to n relations for qso to awards and vice versa... by this way it would also be very easy to have all QSOs for a specific award...

dg9vh, 2020-11-17:
I my eyes after thinking about it, the second way (with having an extra table building up the n:m-relation for qso:awards) would be the best I think.

daccle, 2020-11-17:
For moment I know of five awards, which do use SIG/SIG_INFO:
* COTA
* POTA
* TOTA (Toilets on the Air) (which is a bit specific...)
* WWFF
* GMA (German Mountain Award)

But the SIG/SIG_INFO will produce more awards, since many logging solutions do support the pattern and besides adding MY_SIG/MY_SIG_INFO and SIG/SIG_INFO to adif export logging software does not need to do anything specifically to support even new awards (unlike SOTA/IOTA....).

A good enough reporting would be something like: count all different occurences of SIG_INFO, grouped by SIG and list the sums by SIGs. Then I would get something like you did 20 different spots in WWFF, 10 different spots in COTA, and 42 spots in TOTA (Toilets on the Air).

If in the next step we would allow adif exports based on SIG/MY_SIG we could kill the needs for all the stuff to come.

But still I would like to close this issue here and start a pull request once there is some code to debate on.

closed 2020-12-11 by magicbug