magicbug/Cloudlog #709: Allow users to turn on or off QRZ Real time when saving a QSO in real time

magicbug, 2020-11-17:
While I don't use the QRZ feature, I've had a number of people request being able to return off real-time sending to QRZ when saving a QSO and I fully support this, as it's far better suited to a background task.

However I can appreciate some might like to keep this feature, thus I suggest we make this selectable by the user, this might be better under Station Profile or might spur on #630 on a table for global configuration options

AndreasK79, 2020-11-18:
I'll fix this. I'll put a dropdown to the right of the API Key in the Station Profile. I feel it belongs there, as the API Key is station profile based.

AndreasK79, 2020-11-18:
#712 added.

magicbug, 2020-11-18:
Thanks @AndreasK79 merged

closed 2020-11-18 by magicbug