magicbug/Cloudlog #713: [Object Object] error when attempting to upload QSL image

minorsecond, 2020-11-18:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
When attempting to upload QSL images, an error occurs which states "[Object Object]".

**To Reproduce**
Steps to reproduce the behaviour:
1. Click on a callsign in the logbook
2. Click on QSL Card Management
3. Select cards for upload, & click the upload button
4. See error

**Expected behaviour**
The images should be uploaded without error.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: Laptop
 - OS: macOS
 - Browser: Safari
 - Version 11.0.1

daccle, 2020-11-18:
Please checke the permission on: Cloudlog/assets/qslcard folder.

minorsecond, 2020-11-18:
It's assigned to root:root. I tried changing to root:www-data but got the same result.

daccle, 2020-11-18:
Can you enable logging in config.php. Setting it to 1 will be enough. Then retry and have look into the log?

daccle, 2020-11-18:
BTW: as which user the php is running. I guess it won't be root who runs php as.

minorsecond, 2020-11-18:
I've enabled logging. The log_path variable is set to the default path, but nothing is being written. I suspect this is related to permissions.

PHP is being run by my user, not root. So, I set permissions to myuser:www-data and attempted to upload the image files. No change, and nothing is written to application/logs/ directory.

daccle, 2020-11-18:
Then have log into the servers php log and webservers log. Maybe just restart the php process to enable logging.

daccle, 2020-11-18:
Anyhow we error handling shoul be improved. 

@magicbug I suppose the error handling for the front end is done in footer.php:2348 ff. I tried to get into it, but I am not ease doing changes there. Maybe you want to have a look.

minorsecond, 2020-11-18:
Unfortunately, there are no errors in the error log. I've rebooted the server but the error still persists. 

One question: should the qslcard directory have 775 or 755 permissions set? Or something else? Currently, the group doesn't have write access.

minorsecond, 2020-11-18:
Fixed this. I re-read the wiki and realized that I forgot the `sudo chmod -R g+rw /var/www/html/application/config/` step.

closed 2020-11-18 by minorsecond