magicbug/Cloudlog #721: Improvement to QRZ Logbook Upload Cron

magicbug, 2020-11-26:
Query regarding how the cron works, in theory with this being background work, is it uploading one QSO at a time, or is it bulk uploading, packing up one upload that's then submitted, and if it isn't is there a reason its not? as this would heavily reduce script runtime and database loads.

@AndreasK79 perhaps one for you.

AndreasK79, 2020-11-26:
@magicbug it is uploading one QSO at a time. When I wrote this, I tried several times to feed the API with more than one QSO. I never got it to work. I can recheck the API to see if anything has changed.

magicbug, 2020-11-26:
https://www.qrz.com/docs/logbook/QRZLogbookAPI.html

unless this doc has changed it looks like one at a time still

magicbug, 2020-11-26:
As it's a no go shall close it

closed 2020-11-26 by magicbug