magicbug/Cloudlog #723: Bug: If you edit a qso by doing it via the Live QSO when you click on a past QSO

magicbug, 2020-11-28:
We seem to have a bug when doing the following

- Live QSO
- Past QSO or Last QSO
- Click on Callsign and press Edit
- Save QSO

This seems to create a duplicate QSO rather than edit.

AndreasK79, 2020-11-29:
I tried to reproduce this, but I can't. 

The edit is done as it should. Besides, the javascript function qso_save is called when clicking the save button. This again calls qso_save_ajax in an ajax-call. And in the end it ends up in logbook_model->edit. This only runs update. 

If one were to somehow trigger a save qso at the same time as one saves changes, it would log a new qso, and update the editing qso. I can't really see this happening either. As soon as I enter a call, previous QSOs are listed, but they are not clickable with me.

Have you been able to reproduce this?

magicbug, 2020-11-29:
I have, I'll record my screen so you can see. Could be something I'm doing.

magicbug, 2020-11-30:
Live QSO Window -> Previous Contacts click one -> change Gridsquare save -> duplicate qso.

AndreasK79, 2020-11-30:
Strange. It does not happen here with me.

magicbug, 2020-11-30:
Try doing it after saving the QSO

AndreasK79, 2020-11-30:
That was the missing key. Guess the form is submitted a second time when javascript triggers a page reload.

AndreasK79, 2020-11-30:
It's definitely because of the page reload. If you log a QSO and press F5 to reload page, the QSO is logged again.

daccle, 2020-11-30:
Sorry, I still can't reproduce that bug. I am using Firefox 83.0 (64-Bit) on Windows 10. I can't reproduce, neither on my local docker based dev environment, nor on my hosted production installation.

magicbug, 2020-11-30:
> That was the missing key. Guess the form is submitted a second time when javascript triggers a page reload.

Yep pretty sure that's what it is, I suspect not many people will do what I was, but thought it was worth noting

magicbug, 2020-11-30:
> Sorry, I still can't reproduce that bug. I am using Firefox 83.0 (64-Bit) on Windows 10. I can't reproduce, neither on my local docker based dev environment, nor on my hosted production installation.

See my comments on doing it after saving a qso it's definitely happening I can see it on multiple installs and browsers.

magicbug, 2020-12-02:
Merged thanks!

closed 2020-12-02 by magicbug