magicbug/Cloudlog #727: Improve Documentation in the Wiki

daccle, 2020-11-30:
Logging and Station Profiles had some changes lately. They need more recent screenshots and some description about the available data entry field.

See: https://github.com/magicbug/Cloudlog/wiki/Station-Profiles and https://github.com/magicbug/Cloudlog/wiki/Logging

I am willing to write about the Sig/Sig Info My Sig/My Sig Info parts, but I am pretty under-talented in getting good screenshots and I would like to have some help on all the other available options, which need documentation. Besides I'd love to have a English native-speaker to proof-read my ramblings.

Maybe we can use this Feature request to coordinate our efforts on improving the documentation in the wiki.

- [ ] Get recent good looking screenshots of Logging and Station Profile
- [ ] Write about My Sig and My Sig Info in Station Profile
- [ ] Write about Sig and Sig Info in Loggign
- [ ] Write about the other stuff in Station Profile
- [ ] Write about the other stuff in Logging general tab

AndreasK79, 2020-12-03:
Excellent initiative @daccle. As I usually say: one word is better than none. I've tried to document a little bit when I have added new stuff. I'm not a big writer, so usually it's pretty short.

Just keep up the good work, and add information in the Wiki where you find it's missing.
I'm not a native English speaker, but leave a comment here if you need some proof-reading. I'll take a look when I can.

magicbug, 2020-12-06:
Added a few new screenshots to both sections, although if you want things for certain areas just drop a note with image needed and I'll get It added.

poll-busily, 2020-12-06:
I can also work on this a little bit.

I’d be curious to know your thoughts on alternative documentation frameworks. Here’s an example of something I worked on professionally in the past using Sphinx and ReadTheDocs: https://trustedfirmware-a.readthedocs.io . The source content can be hosted anywhere after being crunched and exported as HTML so it could live in or alongside the Cloudlog install.

magicbug, 2020-12-07:
I've considered dumping the wiki, I was considering VuePress as I've used it for a few client project documentation systems, but at the moment think we will stick for the wiki for now, its markdown files so it can easily move away from it,

magicbug, 2020-12-09:
Now that Github Discussions has started for public repos I'm going to close this thread and suggest it moves to there, I've created a new thread at https://github.com/magicbug/Cloudlog/discussions/741

closed 2020-12-09 by magicbug