magicbug/Cloudlog #733: WAS map view

hairlesshobo, 2020-12-06:
I was wondering if it would be possible for you to add a nap view to the WAS awards page. I envision the map being sort of like the grid map, except that the state would be outlined and shaded either red/yallow/green depending on whether the status is not worked/worked/confirmed. 

Thanks in advance for your time! 

73s 
Steve

AndreasK79, 2020-12-06:
Peter and I discussed this, and it is something we talked about adding. I guess it can go on the wish-list.

magicbug, 2020-12-06:
Yep still on my list of things to add either using Leafetjs or Highcharts (Need to check licencing on this)

magicbug, 2020-12-09:
Possible library found today is https://datamaps.github.io looks quite useful for other ideas of Cloudlog too

hairlesshobo, 2020-12-09:
That library looks awesome! I will have to keep that in mind for some of my future projects too. I could see you making some awesome map views in Cloudlog with that.

Side note, I meant to include this in the original post. I love Cloudlog. You guys are doing a fantastic job on it, please keep it up!!

AndreasK79, 2020-12-09:
@magicbug I think you found the map to use for this :)

closed 2020-12-11 by magicbug