magicbug/Cloudlog #735: Map showing all the years QSOs

magicbug, 2020-12-06:
**_Had this request via email_**

The option to have a map showing QSOs from the year as pins like the /logbook and dashboard maps.

AndreasK79, 2020-12-07:
Perhaps this can be done in a way that you can select a bit what you want to show. I was thinking about something like the form I added for the KML export.

magicbug, 2020-12-07:
Exactly what I was thinking, I like generating the maps for social media posts etc

magicbug, 2020-12-07:
I've actually started this but its got some problems, one the way I'm loading the markers into Leaflet seems to have a long delay with 8000 QSOs then crashes the browser so need to figure a method out to load markers without that happening

The custom option /map/custom works OK but interface needs cleaned up

magicbug, 2020-12-07:
Option might be to do it as a markerCluster (https://github.com/Leaflet/Leaflet.markercluster) but option to ideas.

magicbug, 2020-12-07:
In fact this might be the simple solution https://github.com/eJuke/Leaflet.Canvas-Markers

closed 2020-12-11 by magicbug