magicbug/Cloudlog #740: [LoTW][Sync] Satellite RX Band Not setting

magicbug, 2020-12-09:
Looks like somethings changed in TQSL/LoTW so that satellite rx band isn't being set properly although its being sent and freq is correct.

[image]
[image]

[image]

do4bmw, 2020-12-17:
same counts for QO-100

[image]

https://prnt.sc/w4znmn

magicbug, 2020-12-26:
Worth remembering QO-100 is just a satellite and is handled the same as every other satellite, while this doesn't effect uploaded QSOs matching isn't done on band this has now been fixed, you don't need to reupload anything but going forward should be sorted.

closed 2020-12-26 by magicbug