magicbug/Cloudlog #744: eQSL QSO Upload shows blank QSOs and "Unknown" result

philpem, 2020-12-10:
**Describe the bug**
When uploading QSLs to eQSL, the result is "Unknown". eQSL doesn't seem to update.
[image]

Two completely blank QSOs are shown which I can't find in my log to edit or remove.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'eQSL Import/Export'
2. Click on 'Upload QSOs'
3. Click 'Upload QSOs'
4. See error

**Expected behaviour**
QSOs upload, or a meaningful error message as to why they couldn't be uploaded.

**Screenshots**
Screenshot above.
Can also provide database dump (privately) if this would help.

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser: Firefox
 - Version: 

**Additional context**
I've browsed the database with phpMyAdmin and TABLE_HRD_CONTACTS_V01 has five rows with null callsigns.

magicbug, 2020-12-10:
Thanks definitely a bug, just needs a extra SQL line 👍🏻

magicbug, 2020-12-10:
Now resolved

closed 2020-12-10 by magicbug