magicbug/Cloudlog #745: Editing from eQSL QSO Upload page results in full-screen edit screen with broken tabs

philpem, 2020-12-10:
**Describe the bug**
If I click on a callsign/QSO link from the eQSL QSO Upload screen (Upload tab), the edit window appears full-screen and the tabs along the top don't work.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to  eQSL Export
2. Click on a callsign link (for a QSO to be uploaded)
3. Click on one of the tabs in the edit screen, e.g. eQSO
4. Note that the screen does not update like the other QSO Edit screens.

**Expected behaviour**
QSO edit screen either appears as a popup on the eQSL Upload screen (like other editscreens) -- and tabs work correctly (like other edit screens).

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser Firefox
 - Version 83.0 64bit

magicbug, 2020-12-10:
Thanks in theory they should just show the edit but the info page.. bug will get sorted!

AndreasK79, 2020-12-11:
The dialog popup should spawn here. It seems that one was missed when changes were done a while back.

commit 01e49710 (2020-12-12): [eQSL] Fixed displaying of QSO in table. Fixes #745.
commit 6e98adb7 (2020-12-12): Merge pull request #747 from AndreasK79/eQSL_fix_displayqso

[eQSL] Fixed displaying of QSO in table. Fixes #745.
closed 2020-12-12 by magicbug