magicbug/Cloudlog #757: Maps not displaying anymore

davejhahn, 2020-12-17:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Map no longer displays anything in Logbook or Analytics

**To Reproduce**
Steps to reproduce the behaviour:
Logbook
1. Select Logbook
2. Map is blank

Analytics
1. Select Analytics
2. Select Maps
3. Map is blank

**Expected behaviour**
Map displays log entries

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Brave (Chromium)
 - Version: Version 1.18.70 Chromium: 87.0.4280.101 (Official Build) (64-bit)

**Additional context**
It has been working fine up until the point where I imported a QRZ ADIF file.  It seems like it happened after that.  Previously I was importing WSJT-X ADIF files (something got missed in it, but was in QRZ so I used it to catch that one).

It said all were duplicates so I don't think anything even updated, but after this, map no longer works.  Not sure if something actually did update and changed whatever the map uses?

Must have feature for me!

magicbug, 2020-12-17:
Nothing has changed in the code for those areas, although not sure how old the install is, on most browsers you can hit f12 and see if there's any errors. Other usual suggestion is check with a different browser.

My other other guess it's that there's some hurrendous data in one of the QSOs for gridsquare

Could also try /index.php/logbook/qso_map/25/0 and see if you have data or an error

magicbug, 2020-12-17:
Other thing that might be worth doing is Admin - Update Country Files and get it to recheck all your logbook to make sure QSOs are mapped to countries properly

davejhahn, 2020-12-19:
I was able to reproduce.  I emptied everything in my profile in CloudLog, imported from my ADIF file from JTAlert, everything looks fine (including the maps).  If I export from QRZ and import that file into CloudLog, it indicates success + all of the duplicates that are skipped, but after that, the maps don't work.

In looking at the file, the ADIF version in CloudLog is 3.1.0, the version exported from QRZ is 3.0.4, and the version from JTAlert is 2.2.7.  Not sure if something is getting corrupted when importing the 3.0.4 version from QRZ, but the 2.2.7 version from JTAlert works fine (even if I re-import it, which is what I've been doing up until this one instance).

So, I've reconstructed what I need, but it looks like the QRZ export file has some issue.

dg9vh, 2020-12-19:
Try it with a smaller export if this works. Maybe there are single qsos
causing this effect..

magicbug, 2020-12-26:
Would be interested to see the ADIF from QRZ, the version of ADIF doesn't really matter the changes in ADIF are only todo with extra tags, but this would seem to be a malformed import coming into Cloudlog, feel free to create a discussion and we can try work through QRZs issues but I don't think this is a bug with the code base my end

closed 2020-12-26 by magicbug