magicbug/Cloudlog #760: Band statistics chart showing incorrect number of QSOs

ryandzink, 2020-12-20:
**Describe the bug**
The number of QSOs I am seeing on the Analytics > Statistics pie chart showing # of QSOs per band appears incorrect. I am seeing a much smaller # of QSOs on HF bands than I would expect, and I can see more than that number when searching for contacts on that band.

**Screenshots**
Here is what I see on the analytics page (12 40m QSOs):
[image]

Meanwhile searching for 40m QSOs I see more than that:
[image]

AndreasK79, 2020-12-20:
I just did a test here. The numbers are correct for me. I looked at the query, and it does what it should do.
Do you have several station_profiles with QSOs? The advanced search doesn't distinguish between them. The statistics shows numbers from the active profile.

You can also try to run the query in the database and see what numbers show up:

> SELECT DISTINCT (COL_BAND) AS band, count( * ) AS count 
    FROM table_hrd_contacts_v01
    WHERE station_id = :station_id
    GROUP BY band 
    ORDER BY count desc

Just substitute table name (if changed) and input correct station_id.

ryandzink, 2020-12-20:
@AndreasK79 you're totally right. I mostly operate HF away from my QTH and as soon as I toggled over to a station profile in a different gridsquare I started to see larger numbers of QSOs. I'm going to close this as user error 🤦

closed 2020-12-20 by ryandzink