magicbug/Cloudlog #772: [LoTW] imported QSOs not marked as 'sent'

k7tab, 2020-12-23:
After importing confirmed QSOs from LOTW, using the 'import missing QSOs' option, imported QSOs were not marked as sent to LOTW. This resulted in several thousand QSOs being  re-uploaded to LOTW.

It was so much fun, I did it twice trying to track down a couple of other lotw import issues ;)

magicbug, 2020-12-23:
Couple of questions

Did you use "Mark imported QSOs as uploaded to LoTW" or if you was using an adif did the ADIF have the relevent fields sample would be useful.

k7tab, 2020-12-23:
There is no such option on the LOTW Import page.

In the branch I pulled, the code in LoTW doesn't pass anything to the parameter #4 $markLotw
From Lotw.php line 491/2:
                    if ($station_id != NULL) {
                        $result = $this->logbook_model->import($record, $station_id, NULL, NULL, NULL, NULL);  // Create the Entry

[image]

magicbug, 2020-12-24:
That makes more sense, shall check with mine, can't actually remember the code behind it 😜

commit a503bc7b (2020-12-26): [LoTW] Mark Imported QSOs as sent to LOTW

When importing LoTW QSOs into a fresh cloudlog install mark QSOs as sent to LoTW so they dont get reuploaded.

Thanks to K7TAB for pointing this bug out

Fixes #772
closed 2020-12-26 by magicbug