magicbug/Cloudlog #773: [Lotw] import QSO matching too liberal

k7tab, 2020-12-23:
**Describe the bug**
When importing from Lotw, QSOs are matched a little too easily and will update an existing QSO that shouldn't be matched.

**To Reproduce**
Steps to reproduce the behavior:
1. Work W5ERX on SAT CAS-4B 
2. QSY to CAS-4A where W5ERX will work you again.
3. import these QSOs from LOTW (I did it from an empty db, with 'Import missing QSOs' selected , but suspect it didn't matter)
4. See CAS-4B QSO created and CAS-4A QSO match on 4B QSO & update the 4B QSO

**Expected behavior**
SAT QSO should also match on Satellite name.

**Additional context**
logbook_model->import_check should be checking a little more than timespan/callsign/band, at least, in the case of SAT QSOs.

magicbug, 2020-12-26:
Its pretty liberal to be honest getting it spot on is always a dark art

However don't mind adding a check to the SQL to check a sat name / mode

k7tab, 2020-12-26:
haha Yes, it is a dark art! Thanks! I've got a couple more issues & a question or two to write up. But, first I have to get my LOTW straightened out! I've made quite a mess! I'll get them testable/written up soon! Thanks again!