magicbug/Cloudlog #774: Accumulated statistics not working

ok1cdj, 2020-12-25:
Accumulated statistics not working for me. Latest code from git.

`ERROR - 2020-12-25 12:49:07 --> Query error: Unknown column 'year' in 'where clause' - Invalid query: SELECT date_format(col_time_on, '%Y%m') as year,
                (select count(distinct b.col_dxcc) from TABLE_HRD_CONTACTS_V01 as b where date_format(col_time_on, '%Y%m') <= year and b.station_id = 1) total  from TABLE_HRD_CONTACTS_V01 as a
                      where a.station_id = 1 group by date_format(a.col_time_on, '%Y%m') 
                    order by date_format(a.col_time_on, '%Y%m')
`

AndreasK79, 2020-12-25:
Hmm. It runs fine here, but I guess it has something to do with MySQL version and settings. What version are you running?

ok1cdj, 2020-12-26:
mysql  Ver 14.14 Distrib 5.7.10, for Linux (x86_64)

magicbug, 2020-12-26:
> Hmm. It runs fine here, but I guess it has something to do with MySQL version and settings. What version are you running?

Also not seeing that on min spec mysql-server and also bleeding edge

closed 2021-01-02 by magicbug