magicbug/Cloudlog #779: Logging from GridTracker not working

Vyizis, 2020-12-27:
Using the latest code base, QSO logging from GridTracker is not working, GridTracker shows the connection as 'Passed' and from its logs sends the data but the OSO never appears in the log.

I don't see any errors server side, but looking at the communication with Wireshark gave me the following:

    <h4>A PHP Error was encountered</h4>\n
    \n
    <p>Severity: Warning</p>\n
    <p>Message:  count(): Parameter must be an array or an object that implements Countable</p>\n
    <p>Filename: libraries/Adif_parser.php</p>\n
    <p>Line Number: 187</p>\n

I believe this might be something to do with this change?
https://github.com/magicbug/Cloudlog/commit/0fdce42b333b6aac3a75f8a4d9d643db00c27eef

AndreasK79, 2020-12-27:
Thanks for reporting it. I thought all the bugs were fixed related to the changes in the parser, but obviously not. I'll see if I can figure out what is going on here.

AndreasK79, 2020-12-27:
@Vyizis if you want to fix the problem in your local copy, edit adif_parser.php. After line 119, add this 
`$this->datasplit = preg_split("/<eor>/i", substr($this->data, $this->i));`

I'll push a fix for it.

Vyizis, 2020-12-27:
Have just tested that and its working as expected, thanks!

closed 2020-12-27 by Vyizis

magicbug, 2020-12-27:
Merged PR

commit 339ee6e3 (2020-12-27): [Adif_parser] Bugfix for the feed function that is used by the API. Fixes #779.