magicbug/Cloudlog #793: Update from 1 year old database to last cloudlog - not show any QSO

pa2eon, 2021-01-03:
Just made a new update of the cloudlog software.
First make a copy of all the data AND MySQL database.
Then copy the latest software in the folder .. and copied the config and database.php to the new config folder.

I can login again .. and after that I have to activated the station profile.
When I go to the main page .. I see no old QSO's anymore .. only when I go to the page 'Live QSO' or 'Post QSO' I see the latest QSO in the left screen.
When I click on one of the older QSO's I get a black page with on top a blue line with 'QSO Data' in it .. and no old info of the older QSO.

What do I wrong in this update ?

[image]

[image]

pa2eon, 2021-01-03:
Found the solution !

You have to update the field 'station id' in the database (TABLE_HRD_CONTACTS_V01)  with choice number ( in my case number 1 ) !

May be good to make SQL command to do this in one SQL query.

pa2eon, 2021-01-03:
When you use webmin or other the SQL parm is in one line:
UPDATE TABLE_HRD_CONTACTS_V01 SET station_id = 1

magicbug, 2021-01-03:
Pleased you figured it out.

closed 2021-01-03 by magicbug