magicbug/Cloudlog #803: ADIF import from a TQSL generated ADIF FIle does not work

dl6kbg, 2021-01-09:
[image]

73 Oliver DL6KBG

AndreasK79, 2021-01-09:
Can you show the header and the first qso here? Without that, it is impossible to know what is going on.

dl6kbg, 2021-01-09:
Good morning! Sorry missed that. 

It is only one qso:

----snip----

<CALL:6>CT1BQH
   <BAND:4>13CM
   <MODE:2>CW
   <QSO_DATE:8>20210109
   <TIME_ON:6>000500
   <FREQ:8>2400.019
   <BAND_RX:3>3CM
   <FREQ_RX:9>10489.519
   <PROP_MODE:3>SAT
   <SAT_NAME:6>QO-100
<EOR>

73 Oliver DL6KBG

dl6kbg, 2021-01-09:
Good morning! Here it is:

[image]

AndreasK79, 2021-01-09:
Good morning Oliver!

The problem is that there is no header in the file. The adif-parser does look for a header. It does seem to me that TQSL does not add this,

If you add a `<EOH>` at the top, the file will import.

Andreas.

dl6kbg, 2021-01-09:
Many thanks Andreas! 73 Oliver DL6KBG

um 11:01:

closed 2021-03-25 by magicbug