magicbug/Cloudlog #811: ADIF import uppercase <BAND> field problem.

KeyOz, 2021-01-11:
When importing an ADIF file wich contains an uppercase Bandname like "<BAND:2>2M" then die band in the database will also set to upcase "2M". 
When editing this QSO in Cloudlog the edit field for the band shows "160m" because there is no match of the uppercase data in the database with the dropdown box.

To solve this, convert such fields to lowercase on import.

magicbug, 2021-01-11:
It shouldn't be the code is set to lowercase the band and I cant see this happening in my dev code.

```php
        if(isset($record['band'])) {
                $band = strtolower($record['band']);
        } else {
            if (isset($record['freq'])){
              if($freq != "0") {
                $band = $CI->frequency->GetBand($freq);
              }
            }
        }

        if(isset($record['band_rx'])) {
                $band_rx = strtolower($record['band_rx']);
        } else {
                if (isset($record['freq_rx'])){
                  if($freq != "0") {
                    $band_rx = $CI->frequency->GetBand($freqRX);
                  }
                } else {
                  $band_rx = "";
                }
        }
```

KeyOz, 2021-01-11:
Ok. Sorry. 

It was an issue in an older version. I'm importet this logs some time ago and found that problem today. 

in my backup of that older version this part is:

`
        if(isset($record['band'])) {
                $band = $record['band'];
`
Sorry for that. Thanks.

closed 2021-01-11 by magicbug