magicbug/Cloudlog #828: Duplicated input name in QSO edit

leroydiazg, 2021-01-13:
While working in #827 , I noticed that t both Frequency and Frequency(RX) fields have the same `name` , which could be problematic.

Anyway, I suspect this file is not being in use at this moment, (is it?), as I see that in 'edit_ajax' view they are not the same.

https://github.com/magicbug/Cloudlog/blob/31e2107c98f2aaabf0961d599c541af1e05b501f/application/views/qso/edit.php#L70-L80

magicbug, 2021-01-22:
PR Merged which resolves this

closed 2021-01-22 by magicbug

leroydiazg, 2021-02-08:
I was reviewing the issues I opened here and I've just seen this closed. I think you might have misunderstood me, as what I fixed in #826  was the QSO Labels not this issue with the frequency `name` . 

This issue can still be seen in master.

Anyway, I still think this file is not used, as in edit_ajax.php these names are not the same.

leroydiazg, 2021-02-08:
Thanks @AndreasK79 , it's clear now :)

I didn't see any more issues, but honestly I can't be sure 100%. I noticed this one while @dgdavid and I were trying to fix the QSO form labels and Spanish translations. We haven't looked in depth as we have a lot to learn about ham radio before fully understanding the whole project.

AndreasK79, 2021-02-09:
@leroydiazg thanks for letting us know. And we very much appreciate that you and @dgdavid are contributing to this project.

dgdavid, 2021-02-09:
@AndreasK79, 

Thanks to you too for your kind words as well as for addressing it in #876. From our side, it is not only our aim of contributing to open source projects but also de _pressure_ of having a radio ham father :stuck_out_tongue_winking_eye: :wink: