magicbug/Cloudlog #839: CloudLog import ADIF sets seconds as "00" instead of logged time

davejhahn, 2021-01-17:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
When importing an ADIF file, for example from JTAlert, the QSO times all are "00" for the seconds rather than the time of the QSO.

**To Reproduce**
1. Admin->ADIF Import/Export
2. Select Station Profile, Choose File
3. Upload

**Expected behaviour**
The QSOs have the same time as in the log.  

**Additional context**
This surfaced because I used the LOTW Import feature (not even realizing it was actually sending files to LOTW--I thought it was importing FROM LoTW, maybe "LOTW Import/Export" or something more precise would be helpful.

What I have done up to this point is use TQSL to import my logs entries.   I would then go into CloudLog and use LOTW Import (thinking it was just PULLING the QSOs and setting the Sent/Received flags, not realizing it was also uploading to LOTW).

The problem is there is a duplicate for everyone of my QSOs now.  One, with the correct exact QSO time (e.g. has the seconds), and the other with "00", from CloudLog.

I am hoping this is not going to cause a problem for confirmations--I'm assuming that the LOTW duplicate logic will identify it as a duplicate and match at least one of the two, but not prevent a match.  I've sent a support ticket to ARRL asking how this situation would be handled.

Whether or not LOTW cares about the seconds, not even sure, guessing it does not, but I think that CloudLog should not drop off the seconds entirely because of this.  

At this point, I have to stop using one or the other--until I find if the seconds are important, I'm going to use TQSL for the imports, which is unfortunate, because I'm going to have to manually update all of the LOTW Sent/Received now (for the reports to be correct).

Thanks,

Dave

AndreasK79, 2021-01-18:
Pushed #840. This should take care of the seconds problem.

closed 2021-01-20 by magicbug

davejhahn, 2021-01-20:
Awesome, thank you for the quick response, appreciate it.

Dave