magicbug/Cloudlog #854: [QSO] Auto population of state when you have had a prior QSO

magicbug, 2021-01-27:
Looks like I screwed something in the JSON that grabs call information, if you have had a previous contact the state field isn't populated, this needs fixed.

AndreasK79, 2021-02-08:
You haven't screwed with anything there. As far as I can read the code, state is never fetched. I added it locally, but things are only set if a name is set on the previous QSO.

From logbook.php line 118:
`// Do we have local data for the Callsign?
	if($this->logbook_model->call_name($callsign) != null)`

Usually a name is set, but not always. Perhaps the logic needs a rewrite?

AndreasK79, 2021-02-08:
Wouldn't it be better to just fetch the latest qso based on callsign, ordered by logged time and limit 1? Then fill the array with information? The way it is done now, several calls are made to the database.

commit 42a8657f (2021-02-08): Merge pull request #871 from AndreasK79/master

Fetches state from previous QSO. Fixes #854
commit c35d0d25 (2021-02-08): Fetches state from previous QSO. Fixes #854
closed 2021-02-08 by magicbug