magicbug/Cloudlog #864: Problem adding DOK into DOK-Field

dg9vh, 2021-02-02:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
If trying to add Sonder-DOK 65G13 (valid from 03.09.2019 to 02.09.2020) into the DOK-Field, the entry is deleted, so I can't add this DOK. It is not listed in the "check-list" that pops up.

**To Reproduce**
Steps to reproduce the behaviour:
1. Edit QSO
2. Click on Awards
3. Enter DOK
4. See error

**Expected behaviour**
DOK should be added.

magicbug, 2021-02-02:
This is an interesting problem to have @phl0 do you grab deleted? 

Guess other options just allow deleted to be typed in

magicbug, 2021-02-02:
I'll tag in @AndreasK79 too

dg9vh, 2021-02-02:
Thanks, I think "overloading" the list with manual entries would be a solution... actually we have a nice "autocorrection"-function that automatically upcases all entries, if they are in the list... so you can type q03 which will go to Q03...

This should be persistent beneath all solutions :-)

phl0, 2021-02-02:
We did only include DOKs by now. If special DOKs aka. SDOKs should be included as well we only need a slight change in code. I have a (separate) list of those. Will come up with a patch.

phl0, 2021-02-03:
General question is should we "allow" (S)DOKs that are not on the official list of DARC? Don't see any reason for that other than if Cloudlogs DOK database is out of date though ...

AndreasK79, 2021-02-03:
Thanks for testing @dg9vh. This never crossed my mind, and somehow I thought that you could enter whatever you wanted. I see I was wrong, but selectize has a create function. I tried it out, and with that enabled, it is possible to use values not in the dropdown:
[image]

I'll add this in later.
If you want to test, edit footer.php and add
`create: function(input) {
            return {
                value: input,
                name: input
            }
        },`

to this function
`$('#darc_dok').selectize({`

phl0, 2021-02-03:
That would be one option. But still the question: Does it make sense to add "arbitrary" (S)DOKs or should we rely solely on the list by DARC (+ plus maybe "NM" for "not member" sometimes needed during contests)?

AndreasK79, 2021-02-03:
@phl0 I didn't answer, because I don't know. Are they official, and valid for any award?

phl0, 2021-02-03:
No problem at all. I just parse the official info by DARC on a regular basis. So yes that would be the "official" list and only those would be valid for the German awards.

dg9vh, 2021-02-03:
@phl0 my opinion is only to list officials but also those, who are out of valid time. Did not check DARC list of sdok for the mentioned one, but there was a qsl card with official stamp for the sdok so I thought it was correct.

So the Sonder-DOK 65G13 is not officially issued you say?

phl0, 2021-02-03:
Nope. My list contains all SDOKs that we assigned once (and my be not valid today). See https://www.df2et.de/cqrlog/sdok.txt

dg9vh, 2021-02-03:
So I will check validity against the official site...

phl0, 2021-02-03:
You don't need to. I will patch @AndreasK79's code so that it also pulls the SDOK data. You are the test bunny then.

dg9vh, 2021-02-03:
Ok. Std-by

dg9vh, 2021-02-03:
To get a list with official issued sdok visit https://www.darc.de/fileadmin/filemounts/referate/sdok/SDOK_List.csv

dg9vh, 2021-02-03:
> That would be one option. But still the question: Does it make sense to add "arbitrary" (S)DOKs or should we rely solely on the list by DARC (+ plus maybe "NM" for "not member" sometimes needed during contests)?

I forgot adding NM for contests would be a fine solution :-)

phl0, 2021-02-03:
Guys I did a quick and dirty implementation on my DOK and SDOK parser. It pulls the official lists from DARC.de and parses them for CQRlog and also provides dok.txt and sdok.txt on my webspace. I changed the script so that it produces also dok_and_sdok.txt so that this file contains dok.txt, "NM" and sdok.txt in a single file.
My code is here: https://github.com/magicbug/Cloudlog/compare/master...phl0:AddSDOKinfo
Happy to hear your test results before I PR this.
@magicbug I also added assets/json/* to .gitignore in order to not get confused by (not) updated contents here. Is that ok or should I limit that to the dok.txt (and maybe sota.txt)?

phl0, 2021-02-04:
Added some more minor changes to my AddSDOKinfo branch...

magicbug, 2021-02-08:
Merged into main branch thanks

closed 2021-02-08 by magicbug