magicbug/Cloudlog #866: Lightbox not showing map

PD0RZH, 2021-02-05:
In [my setup](http://pd0rzh.beukie.com/cloudlog) the lightbox (after clicking a callsign from homescreen)  is not showing a map. I'm not sure whether it has worked before but if it did then I have a bug... ;-)

Any ideas?

magicbug, 2021-02-05:
If your using a modern desktop browsers hit f12 then go to console in the developer tools then refresh the page and click again to see what errors come up.

I'm going to guess your using the latest codebase at this stage

PD0RZH, 2021-02-05:
[image]
This is what it comes up with

Regards Rob

magicbug, 2021-02-05:
I'm guessing in that case you aren't running things using Git in which case latest code base can be had via https://github.com/magicbug/Cloudlog/archive/master.zip

magicbug, 2021-02-05:
I've a feeling if you update your code this will go away

PD0RZH, 2021-02-05:
Your feeling did not let you down. Thanks Peter

closed 2021-02-05 by PD0RZH