magicbug/Cloudlog #873: Non-fatal PHP warnings

ofadam, 2021-02-08:
Since upgrading to PHP 8 on my self-hosted Cloudlog instance, I occasionally get the following deprecated warnings. Everything seems to function as usual, but just thought I'd mention in case you haven't jumped to 8.

Deprecated: Required parameter $skipDuplicate follows optional parameter $station_id in application/models/Logbook_model.php on line 1494

Deprecated: Required parameter $markLotw follows optional parameter $station_id in application/models/Logbook_model.php on line 1494

Deprecated: Required parameter $dxccAdif follows optional parameter $station_id in application/models/Logbook_model.php on line 1494

Deprecated: Required parameter $markQrz follows optional parameter $station_id in application/models/Logbook_model.php on line 1494

magicbug, 2021-02-08:
I'm actually surprised it works, I wasn't expecting this version of Codeigniter to actually work with php8, I'm sure the warnings can be addressed

ofadam, 2021-02-08:
It was an ill-advised upgrade, no doubt!

commit 4ca443de (2021-02-08): Merge pull request #877 from AndreasK79/php8_warning_fixes

This fixes the PHP8 warnings. Fixed #873
commit f122a284 (2021-02-08): This fixes the PHP8 warnings. Fixed #873
closed 2021-02-08 by magicbug