magicbug/Cloudlog #880: Possible problem with translations and selected default language

dgdavid, 2021-02-09:
:warning: DISCLAIMER: maybe what I'm going to report is completely wrong, but I strongly believe it worth having a look at it :warning: 

## Context

Last weekend, @leroydiazg pinged me because he noticed that something weird was happening with the Spanish translations in the instance we have set up. That instance has the 'spanish' language set as default in the config file ([application/config/config.php](https://github.com/magicbug/Cloudlog/blob/4ca443de3f831d3e833fb7a5cd3b41c71252a80a/application/config/config.sample.php#L158)).

However, Leroy told me that "Notes" translations were not working as expected despite the 'application/languages/spanish/notes_lang.php' was there while _the rest_ of the translations were being shown. Mysterious enough.

## The problem (?)

So, after looking here and there to check everything was fine, we ended doing a rudimentary debugging<sup>[1]</sup> of the [system/core/Lang.php file](https://github.com/magicbug/Cloudlog/blob/4ca443de3f831d3e833fb7a5cd3b41c71252a80a/system/core/Lang.php), using the PHP `print` command inside the `load` method :man_facepalming: to see which _idiom_ was being used per each _langfile_. The result was something like 

```
general_words_lang.php => 'spanish'
migration_lang.php => 'spanish'
notes_lang.php => 'english'

```

That pointed us out to the migration files and... there seems to be _half of the problem_<sup>[2]</sup>:

https://github.com/magicbug/Cloudlog/blob/4ca443de3f831d3e833fb7a5cd3b41c71252a80a/application/migrations/061_add_lang_to_options.php#L15

As far as we can see, support for setting the language preference in the database [has been added recently](https://github.com/magicbug/Cloudlog/commit/d2aec5d15952eca9dcf173a49276d6f6c19699dd), setting it to 'english' by default.

So, looks like [autoloaded lang files](https://github.com/magicbug/Cloudlog/blob/4ca443de3f831d3e833fb7a5cd3b41c71252a80a/application/config/autoload.php#L99) ('general_words' at this moment) are being loaded using the default language set in the config file and the rest using the preferred language set by the DB.

## Proposed solutions (or just crazy ideas)

Assuming that I'm right (which could be not), I'd propose

* Not forcing the preferred language to be 'english'. Instead, leave it blank meaning to use the default language set in the config file
* To reload (if such thing is possible) the autoloaded languages when changing the preferred language

Or (even better?) to wrap `$this->load->lang` in a custom helper to set the preferred or default language as the second param when it is not given.

---

<sub>[1] Probably looking at server logs would have been a better idea, but that's another story.</sub>
<sub>[2] The other half, maybe, it's in the `OptionsLib` - https://github.com/magicbug/Cloudlog/commit/4770340c45321f7c1d081e9c61b052210076194a

magicbug, 2021-02-09:
What I did do was write some frontend code that added language selection inside global options it's then grabbed and set with all the other options data.

Ultimately this should overide the. Config.php selection and you'd need to reselect within the frontend but I didn't expect that to cause any problems

dgdavid, 2021-02-09:
> but I didn't expect that to cause any problems

It does, I just checked it right now by setting the instance in the other way around: 'english' in the config file, 'spanish' in the database. As said, `general_words` lang file (the one autoloaded) is honoring the configuration set in the config.php file. Yo can do a simple test by yourself or maybe one of the other people involved in other translations can do it and confirm or deny it.

dgdavid, 2021-02-09:
@ok1cdj, @oh1jw, @phl0, friendly ping :point_up: :point_up:

magicbug, 2021-02-09:
OK so there does appear to be an issue with languages which don't have appropriate code lang files present in Codeigniter like Finnish you get errors like

Unable to load the requested language file: language/finnish/pagination_lang.php

It makes me think that these files need to be included else nothing works

Looking at the QSO Panel for each
- Czech seems complete
- Dutch - seems complete
- English - fine
- finnish - get errors as above
- french - loads ok
- German for example if i go to QSO I get lots of missing fields
- Spanish loads OK

this was with changing languages within global options and the config.php set to $config['language']	= 'english'; 

So I'm confused by what the problem is although bit drugged up on meds, German I thought was OK although doesn't seem to be so I'll fall back to @phl0 to have a look, Finnish it seems we need a core set of lang files to handle things correctly.

thus I'll wait to see what the others say, as I'm not quite grasping the issue.

magicbug, 2021-02-09:
If you want to test what I changed you can comment out `$CI->config->set_item($item->option_name, $item->option_value);` from libraries/OptionsLib.php on line 30.

magicbug, 2021-02-09:
OK so Unable to load the requested language file: language/dutch/notes_lang.php doesn't load if its missing and removing the code changes I made makes zero change once a views been converted there's no fallback if a language to english

dgdavid, 2021-02-09:
> OK so there does appear to be an issue with languages which don't have appropriate code lang files present in Codeigniter like Finnish you get errors like
>
> Unable to load the requested language [...]

Go to notes instead. Most probably you'll see nothing translated.

> So I'm confused by what the problem is although bit drugged up on meds, German I thought was OK although doesn't seem to be so I'll fall back to @phl0 to have a look, Finnish it seems we need a core [...]

Sorry if I was not able to express the issue clearly. I tried it, believe me.

dgdavid, 2021-02-09:
> If you want to test what I changed you can comment out `$CI->config->set_item($item->option_name, $item->option_value);` from libraries/OptionsLib.php on line 36.

In fact, here is where I think the problem starts. Commenting the line, the language set in the config file is always honored. When not, the translations declared in the _general\_words_lang are being translated to the language set in the config file and the rest (or at least those in _notes\_lang_) uses the language set by this line.

At least in my tests.

magicbug, 2021-02-09:
Just to quickly resolve this till I get more hours in the day, I've removed the option and hidden the feature in global settings for now config manual change will be the only option.

closed 2021-02-09 by magicbug

oh1jw, 2021-02-27:
I noticed that there was completely missing finnish translation at Codeigniter. I just translated it and made PR. 
It might solve the problem.

dgdavid, 2021-02-28:
> I noticed that there was completely missing finnish translation at Codeigniter. I just translated it and made PR.

Great! Don't forget to add them here too, maybe once [your PR](https://github.com/bcit-ci/codeigniter3-translations/pull/432) get merged.

> It might solve the problem.

Actually not. Well, let's say _partially_. I mean, the problem spotted here is different that the one [I already mentioned in](https://github.com/magicbug/Cloudlog/pull/799) related to the "empty" string when translation is missing, which most probably is what you noticed.

Anyway, Peter already undid the changes related with this issue.

oh1jw, 2021-02-28:
> > I noticed that there was completely missing finnish translation at Codeigniter. I just translated it and made PR.
> 
> Great! Don't forget to add them here too, maybe once [your PR](https://github.com/bcit-ci/codeigniter3-translations/pull/432) get merged.
> 
Sorry, I'm a newbie with Github - is there any easy way to duplicate [them](https://github.com/oh1jw/codeigniter3-translations/tree/develop-finnish/language/finnish) after they are merged?

dgdavid, 2021-03-01:
> Sorry, I'm a newbie with Github

No problem, welcome :)

> - is there any easy way to duplicate [them](https://github.com/oh1jw/codeigniter3-translations/tree/develop-finnish/language/finnish) after they are merged?

I don't think so<sup>[1]</sup>. Just follow the https://github.com/bcit-ci/codeigniter3-translations instructions and 

> Copy the folder(s) for the idioms you are interested in, from inside the language folder of this project to your application/language folder.

and made a PR against Cloudlog

---

<sub><sup>[1]</sup> although the project could use a [Github action](https://docs.github.com/en/actions) or other process/technique to automatize the task of fetching the most recent translations for languages at application/languages from https://github.com/bcit-ci/codeigniter3-translations in, let's say, each release. But definitely this is nothing you can do and it is not "the easy way" you were asking for :)</sub>

dgdavid, 2023-05-15:
JFTR, fixed by @AndreasK79 in https://github.com/magicbug/Cloudlog/pull/2138