magicbug/Cloudlog #890: Error at line 163 and 169 of Adif_parser.php when downloading QSOs from eQSL

lu1aax, 2021-02-17:
**Describe the bug**
This is the text I get every time I import QSOs from eQSL

A PHP Error was encountered
Severity: Notice

Message: Uninitialized string offset: 186

Filename: libraries/Adif_parser.php

Line Number: 163

A PHP Error was encountered
Severity: Notice

Message: Uninitialized string offset: 186

Filename: libraries/Adif_parser.php

Line Number: 169

**To Reproduce**
Download QSOs from eQSL

**Expected behaviour**
No error

**Additional context**
This happens since I installed Cloudlog at my server

magicbug, 2021-02-17:
Before we can troubleshoot ADIF files we require a copy of it this will be via exporting it via eQSLs website this will allow us to identify the issue but without it its hard todo as in my live copy I don't get this error

lu1aax, 2021-02-18:
Here it goes

[LF1AAC36450.adi.txt](https://github.com/magicbug/Cloudlog/files/6000325/LF1AAC36450.adi.txt)

AndreasK79, 2021-02-18:
Hi.

Thanks. That made things much easier. This line is the problem:
`<CALL:9>PY2WLM   <QSO_DATE:8:D>20201231<TIME_ON:4>0022<BAND:3>10M<MODE:3>FT8<RST_SENT:3>-09<RST_RCVD:0><QSL_SENT:1>Y<QSL_SENT_VIA:1>E<APP_EQSL_AG:1>Y<GRIDSQUARE:6>gg66rm<EOR>`

The call tag says 9 characters,, but are only 6. There are 3 spaces after the call. When I fixed that, everything worked ok.

magicbug, 2021-02-20:
Closing this as it's a eQSL issue and not a Cloudlog one

closed 2021-02-20 by magicbug