magicbug/Cloudlog #893: Unable to create station profile due to missing field dxcc

hb9gix, 2021-02-22:
**Describe the bug**
It's currently not possible to create a station profile (or edit one), due to the field dxcc (or station_country) being missing from the form

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Admin > Station Profiles > + Create Station Profile
2. Fill out all the required fields
3. Klick 'Create Station Pofile' button on the bottom of the page
4. See error ""

**Expected behaviour**
Station Profile is created

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Linux (Kubuntu 20.04)
 - Browser: Firefox 
 - Version 85.0.1 (64-bit)

**Additional context**
The issue can be remedied by inspecting the html code and inserting the form field "dxcc" with the appropriate value and setting the value of the form field "station_country" manually

magicbug, 2021-02-22:
Missed the step to update country files "admin" - "Update Country files"

hb9gix, 2021-02-22:
Point taken.
But shouldn't it be handled nontheless?

magicbug, 2021-02-22:
New install procedure is in the works which will force all the relevant steps to be carried out, we don't package the country files as the data changes on a regular basis with exceptions lists for DX trips (not such an issue with covid!)

closed 2021-02-22 by magicbug