magicbug/Cloudlog #923: Old DXCC tables should be removed

wvolz, 2021-03-05:
**Describe the bug**
Two old DXCC tables exist and I want to confirm these are good for removal. I'll submit a pull request with migration to remove these if ok.

**To Reproduce**
Look for tables 'dxcc' and 'dxccexceptions'.

**Expected behaviour**
The code is now using 'dxcc_entities' and 'dxcc_exceptions'.

**Additional context**
I noticed this after updating from a 2019 build to the Jan 2021 (or there about).

magicbug, 2021-03-09:
Current DXCC tables are

dxcc_entities
dxcc_exceptions
dxcc_prefixes

You should be able to delete dxcc and dxccexceptions safely

closed 2021-03-09 by magicbug

wvolz, 2021-04-05:
The old table names were still present in the code in a couple of places. Please see pull request #982 to fix this.