magicbug/Cloudlog #938: [Possible bug] PHP error when posting a live qso

kitobelix, 2021-03-14:
Hello!! I have recently seen a possible bug in cloudlog (only) when posting a live qso. On the very top of the page i see two errors related with the headers. This happens on every browser I tryed, from mobile chrome to firefox desktop.

`A PHP Error was encountered
Severity: Warning

Message: Cannot modify header information - headers already sent by (output started at /cloudlog/application/models/Logbook_model.php:2305)

Filename: controllers/Qso.php

Line Number: 87`

`A PHP Error was encountered
Severity: Warning

Message: Cannot modify header information - headers already sent by (output started at /cloudlog/application/models/Logbook_model.php:2305)

Filename: controllers/Qso.php

Line Number: 88`

Unfortunately I had no time to spend trying to find a possible solution. I also tryed modifying the php version but that doesn't helped at all.

Also I found this discussion https://github.com/magicbug/Cloudlog/discussions/884 but there is no solutions proposed.

How can I contribute to the fix?

Thank you very much!

magicbug, 2021-03-14:
Can you let us know the following

- PHP version
- Whether you have QRZ Logbook enabled on the current station profile.

magicbug, 2021-03-14:
Think the solution will be to change the two set cookies to session items I'll try get this done this afternoon

magicbug, 2021-03-14:
I'm not sure if https://github.com/magicbug/Cloudlog/commit/341c014a8e76b5d631709ae24362cc83706b4f6 will fix this issue but would be interested to know if it made any difference

kitobelix, 2021-03-22:
i! Sorry I was bloated of work lately. I'm using php 7.0 since i saw the error. Previously on 7.4 with equal results. Seems like something trivial to solve. Thank you very much!

kitobelix, 2021-03-22:
I just had another situation, possibly related.

I was editing a station. I've found that one of my stations had a wrong Country selected. When I edited it, got this:

> A PHP Error was encountered
Severity: Warning

> Message: Cannot modify header information - headers already sent by (output started at /srv/users/serverpilot/apps/cloudlog/public/application/models/Logbook_model.php:2312)

> Filename: helpers/url_helper.php

> Line Number: 564

Yet, my station details were updated successfully.

kitobelix, 2021-03-22:
I just had another situation, possibly related.

I was editing a station. I've found that one of my stations had a wrong Country selected. When I edited it, got this:

> A PHP Error was encountered
Severity: Warning

> Message: Cannot modify header information - headers already sent by (output started at /cloudlog/public/application/models/Logbook_model.php:2312)

> Filename: helpers/url_helper.php

> Line Number: 564

Yet, my station details were updated successfully.

magicbug, 2021-03-22:
Don't worry know the issues of work sucking time! I suspect the other issues need similar treatment will check those, 

I'm guessing the commit solved the qso page error so hopefully others will be something similar, although I don't think station profiles use cookies.

magicbug, 2021-03-24:
> > A PHP Error was encountered
> > Severity: Warning
> > Message: Cannot modify header information - headers already sent by (output started at /cloudlog/public/application/models/Logbook_model.php:2312)
> > Filename: helpers/url_helper.php

This was caused by whitespace in the file should be resolved too

commit 341c014a (2021-03-14): Moved Cookie set to before QSO Save

This might fix #938
closed 2021-03-24 by magicbug