magicbug/Cloudlog #943: qrz.com callbook lookup problem

atremar, 2021-03-15:
Logging a LiveQSO with 'qrz' configured as callbook (config.php) nothing happens.
QRZ credentials and subscription are valid. hamqth works.

_DEBUG - 2021-03-15 07:31:59 --> Qrz class already loaded. Second attempt ignored._

Cloudlog: fresh install, latest version (nightly update_cloudlog.sh)
PHP: v7.4.16
Browser on MacOS: Safari, Firefox

Thanks a lot for any further help or idea ...
73 de Meno, HB9HID

magicbug, 2021-03-16:
OK very weird, so that is happening even with the following items 

$config['callbook'] = "qrz";
$config['qrz_username'] = "";
$config['qrz_password'] = "";

configured in /application/config/config.php? once you have configured it, you might need to logout then login just to be sure but I honestly cant think what would cause it not to work, happy to ponder some more on it however

atremar, 2021-03-17:
Hi Peter,
TNX for checking ... 

The same Cloudlog codebase and config working perfectly with QRZ on other environments.

It's definitely a problem with the Webhosting provider's environment.
Will not waste time there with limited possibilities and move on..

Cheers and 73

closed 2021-03-17 by atremar