magicbug/Cloudlog #960: Awards: SIG - doesn't return anything

oh1jw, 2021-03-20:
**Describe the bug**
Awards: SIG - doesn't return anything at http://cloudlog-intance/html/index.php**/awards/sig**

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to **Awards**
2. Click on **SIG**
3. See error

**Expected behaviour**
I think there should be way to choose what kind of information you are seeking at SIG-field. 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: OS X / Windows 10
 - Browser: Chrome
 - Version: latest

AndreasK79, 2021-03-20:
The SIG award looks for anything in the COL_SIG column in the database. If you have not put anything in that column (SIG) while logging the QSO, nothing will be found by the award.
[image]

oh1jw, 2021-03-20:
I did a double check and I do have a SIG notation done on more than one qso. I also checked with different station profiles.

oh1jw, 2021-03-20:
Ok - it works only new qsos, made after the new feature.

AndreasK79, 2021-03-20:
It shouldn't matter if the QSO is new or old, as long as SIG contains a value and SIG info contains a value.

oh1jw, 2021-03-20:
I managed to get it show something, when I logged new WWFF-qso. After that, the SIG-award will result that just logged qso. 
[image]

AndreasK79, 2021-03-20:
If you edit the OH1MM/P qso, what is written in SIG and SIG INFO?
[image]

oh1jw, 2021-03-20:
[image]

[image]

oh1jw, 2021-03-20:
I got it to work as it should when I made a small side jump: I activated one more drive profile and went back original. 
Now it shows previous WWFF contacts as it should.

I’ll see how it goes when that my “official” Cloudlog instance, hosted on the Cloudlog server gets an update at midnight.

AndreasK79, 2021-03-20:
I think you found the issue then. Only QSOs with the active station profile is shown. I guess if no profile is active, then "nothing found".

oh1jw, 2021-03-20:
> I think you found the issue then. Only QSOs with the active station profile is shown. I guess if no profile is active, then "nothing found".

To clarify the path: 
1. "Nothing found" with active station profile OH1JW
2. I made one wwff-qso --> 1 result, that just made qso. 
3. Side jump to station profile OH1BAR
4. "Nothing found"
5. Back to station profile OH1JW --> many results with WWFF. 
6. Chance to station profile OH1BAR**/m**, "nothing found" - but I know there should be few WWFF qsos.

AndreasK79, 2021-03-20:
Thanks for the clarification. Then I'm not sure what is going on here. @magicbug any ideas here?

magicbug, 2021-03-20:
I'd be interested to know if it's the same on the hosted copy, running this code fine on my live copy since I merged it.

oh1jw, 2021-03-20:
> I'd be interested to know if it's the same on the hosted copy, running this code fine on my live copy since I merged it.

Yep - will see. I won't do any changes to that hosted copy, so you can see it yourself if the problem occurs there. 
I'm not sure, but it might be something related to my multiple station profiles. Or not. Well see. 
My local Cloudlog for testing is running at Pi3 with DietPi (MariaDB).

oh1jw, 2021-03-21:
At a hosted environment at Magicbug I did not get the same problem repeated. 
I will continue my testing locally and I will document the process if I manage to trigger it again. 
Thank you for your tima and effort.

--> **I will close this issue report for now.**

closed 2021-03-21 by oh1jw