magicbug/Cloudlog #966: PHP Error - Message: Module 'sqlite3' already loaded

ben-kuhn, 2021-03-26:
Hello,

I am installing/testing the latest Cloudlog from git as of 26 Mar, 2021 on a qth.com shared hosting instance for our local club's logs.  I am encountering the following error on every page post-install:

`A PHP Error was encountered
Severity: Core Warning

Message: Module 'sqlite3' already loaded

Filename: Unknown

Line Number: 0`

I have limited access to the system, so changing PHP modules is not something I am able to do.  Here is the PHP version information:

`php --version
PHP 7.4.15 (cli) (built: Feb  4 2021 22:33:12) ( NTS )
Copyright (c) The PHP Group
Zend Engine v3.4.0, Copyright (c) Zend Technologies
    with Zend OPcache v7.4.15, Copyright (c), by Zend Technologies
`
The shared host appears to be unix-like running Apache.

magicbug, 2021-03-26:
Hi Ben,

Not sure why you are getting sqlite3 errors, as this isn't a supported database type for Cloudlog nor used by Cloudlog

Peter.

magicbug, 2021-03-26:
If the page is loading ok apart from that error you can edit index.php in the main folder and change `define('ENVIRONMENT', 'development');` and change it to` define('ENVIRONMENT', 'production');`

That being said, not sure the about the other error i'd suggest talking to your hosting provider.

ben-kuhn, 2021-03-26:
Hi Peter,

Everything is up and running correctly with mysql, so I really have no idea why it would complain about sqlite.

The error is completely superficial so it's not causing any real problems.  I was able to import an ADIF and use the application as expected.  Changing 'ENVIRONMENT' to production suppressed the error so good enough for me.  Thanks for your assistance.

Ben

closed 2021-03-26 by ben-kuhn