magicbug/Cloudlog #978: Upload to Clublog fails with "operation aborted by callbackError"

exitnode, 2021-03-31:
**Describe the bug**
I can't upload to clublog (qrz, eqsl and lotw are working). When I try, I get the error "operation aborted by callbackError"

**To Reproduce**
Steps to reproduce the behaviour:

1. curl --silent https://log.qrz.is/index.php/clublog/upload/dl6mhc

**Expected behaviour**

Upload of ADIF data to clublog

**Additional context**

I activated logging on the application level. Here is the output:

```
INFO - 2021-03-31 12:49:52 --> Model Class Initialized
INFO - 2021-03-31 12:49:52 --> File loaded: /home/micha/websites/log.qrz.is/htdocs/application/views/adif/data/clublog.php
ERROR - 2021-03-31 12:49:52 --> Clublog upload for DL6MHC failed reason
INFO - 2021-03-31 12:49:52 --> Final output sent to browser
DEBUG - 2021-03-31 12:49:52 --> Total execution time: 0.4009
```

Application has been updated via git -> no change
I changed the password of my clublog account to not have any funny characters -> no change
I tried entering my clublog mail address instead of my callsign (dl6mhc and DL6MHC) in config dialog -> no change

Here is what the web server log says:
`[31/Mar/2021:15:14:10 +0200] "GET /index.php/clublog/upload/dl6mhc HTTP/2.0" 200 35 "-" "curl/7.74.0"`

magicbug, 2021-03-31:
The error `operation aborted by callback` is likely coming out of curl which makes me think PHP isn't able to create files within /uploads directory we do this to make a temporary.adi file that's sent to Clublog

exitnode, 2021-03-31:
Thanks for your response. I made the uploads directory temporarily world writeable but this didn't change the behaviour. Is there any way I can debug this?

exitnode, 2021-03-31:
It works now. The problem was that I configured the variable _$this->config->item('directory')_ inside _config.php_ with the full path to the installation folder. Because of this, the following if-statement decided to choose the first case:

```
if($this->config->item('directory') != "") {
    $filepath = $_SERVER['DOCUMENT_ROOT']."/".$this->config->item('directory')."/".$file_info['server_path'];
} else {
    $filepath = $_SERVER['DOCUMENT_ROOT']."/".$file_info['server_path'];
}
```

This resulted in _$filepath_ to be constructed as:

`/home/micha/websites/log.qrz.is/htdocs//home/micha/websites/log.qrz.is/htdocs/uploads/clublogxxxxxxxxxxx.adi`

I seem to have misunderstood the purpose of this configuration item.

magicbug, 2021-04-02:
Pleased its resolved

closed 2021-04-02 by magicbug