magicbug/Cloudlog #980: Error on manual synch to LoTW

Vic152, 2021-04-02:
I love the bug template but this is fairly straightforward ;-)

This happens when I do manual synch to LoTW.

I am not sure which version of the Cloudlog we run at https://cloudlog.oarc.uk/

[image]

magicbug, 2021-04-02:
Is /uploads/lotw folder writable? First error brings that into question

Vic152, 2021-04-02:
> Is /uploads/lotw folder writable? First error brings that into question

Will check.

magicbug, 2021-04-02:
PHP needs read/write access as per the install doc on the wiki on certain folders

Vic152, 2021-04-04:
Permissions updated all fine! Thanks for the tip! Well done on the software. I really like it. It's my main logbook.

closed 2021-04-04 by Vic152