magicbug/Cloudlog #981: [Contesting] Contest Logbook not restoring

magicbug, 2021-04-05:
@AndreasK79 Not sure if I've changed something or just missing or forgot how it worked but noticed within Contesting after adding a few QSOs and leaving the area and going back the Contest Logbook is no longer restoring like I think it was before.

magicbug, 2021-04-05:
@AndreasK79 in fact just did some testing and called the `restoreContestSession()` function within the browser developer tools console and this was returned so guess that's probably the issue?

DataTables warning: table id=DataTables_Table_0 - Cannot reinitialise DataTable. For more information about this error, please see http://datatables.net/tn/3

magicbug, 2021-04-05:
@AndreasK79 Looks like I've resolved this, if you can confirm that I've done it right within the contesting section please :)

AndreasK79, 2021-04-05:
@magicbug that is strange. I just tried to log a QSO here in contest mode. The QSO was restored when I came back. I have all the commits, except the last one with the fix.

magicbug, 2021-04-05:
Weird! Put it down to unknown then 🙂

closed 2021-04-05 by magicbug

AndreasK79, 2021-04-05:
Sure is, but I guess a check to see if it is initialized is good anyway. I haven't touched the code here in a long time, but you never know when other changes might interfere.