magicbug/Cloudlog #983: Add QSO - retains data from previous satellite QSOs when logging non-satellite QSOs

timkirby, 2021-04-07:
I use the Add QSO feature to enter satellite QSOs into the log. I set the satellite name and satellite mode on the sat tab and save the QSO. All good. 

If I then want to add a non-satellite QSO, say 80m CW, before I log it, I have to remember to go to the satellite tab and blank out the satellite name and satellite mode fields, and then go to the General tab and change the 'Propagation Mode' field from 'Satellite' to blank.

If I do not, in the case of our 80m QSO example, despite the band field showing 80m, it gets logged as, AO-91. And if I don't change the 'Propagation Mode' field, on uploading to LOTW, the 80m QSO will show as 'Satellite, Internet or Repeater'.

Would it be possible to change the functionality, perhaps so that if the satellite name is set to blank, it clears both the satellite mode and propagation mode fields?

I hope this makes sense! As ever - thank you for all your work.

Tim GW4VXE

AndreasK79, 2021-04-10:
Hi Tim.

This is doable. I can take a look when I find some time.

timkirby, 2021-04-10:
Hi Andreas - that’s very kind of you! Thank you!

Tim

AndreasK79, 2021-09-20:
This is merged in station_logbooks branch. When this is done, it will be merged into the master branch.

commit 15e42355 (2021-09-16): [QSO entry] Blanking satellite blanks mode and propagation mode. Fixes #983
closed 2021-09-20 by AndreasK79